mkovero/7226ctrl #4: Create hardware test-mode

mkovero, 2018-06-06:
Idea is to create simple test mode for controller to see if wiring is alright.
- test sensors / solenoids / actuators (test that tps is in range, if not, calibrate it etc.)
- display test / align (support for different displays people might have)

dvjcodec, 2018-11-19:
I use simple sketch for tests outputs in my controller. Just add LCD output with test graphics. 
https://drive.google.com/file/d/1jQX9WTmhtOgBZNdRIwTGqzLBBpeP_FXw/view?usp=sharing

int led = 19;   //speed1
int led2 = 18;  //speed2
int led3 = 24;  //speed3
int led4 = 35;  //selenoid1
int led5 = 37;  //selenoid2
int led6 = 41;   //selenoid3
int led7 = 43;  //selenoid4
int led8 = 39;   //selenoid5
int led9 = 29;   //selenoid6
int led10 = 31;   //selenoid7
int led11 = 33;   //selenoid8
int led12 = 2;  //MCP int
int led13 = 0;   //
int led14 = 0;  //
int led15 = 0;  //
int led16 = 0;  //
int led17 = 0;  //
int led18 = 0;  //
int led19 = 0;  //

void setup() {               
// initialize the digital pin as an output.
pinMode(led, OUTPUT);
pinMode(led2, OUTPUT);
pinMode(led3, OUTPUT); 
pinMode(led4, OUTPUT);
pinMode(led5, OUTPUT);
pinMode(led6, OUTPUT);
pinMode(led7, OUTPUT);
pinMode(led8, OUTPUT);
pinMode(led9, OUTPUT);
pinMode(led10, OUTPUT);
pinMode(led11, OUTPUT);
pinMode(led12, OUTPUT);
pinMode(led13, OUTPUT);
pinMode(led14, OUTPUT);
pinMode(led15, OUTPUT);
pinMode(led16, OUTPUT);
pinMode(led17, OUTPUT);
pinMode(led18, OUTPUT);
pinMode(led19, OUTPUT);
}

void loop() 
{digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
delay(50);                 // wait for a second
digitalWrite(led, LOW);     // turn the LED off by making the voltage LOW
delay(50);                 // wait for a second

{digitalWrite(led2, HIGH);
delay(50);
digitalWrite(led2, LOW);
delay(50);}

{digitalWrite(led3, HIGH);
delay(50);
digitalWrite(led3, LOW);
delay(50);}

{digitalWrite(led4, HIGH);
delay(50);
digitalWrite(led4, LOW);
delay(50);}

{digitalWrite(led5, HIGH);
delay(50);
digitalWrite(led5, LOW);
delay(50);}

{digitalWrite(led6, HIGH);
delay(50);
digitalWrite(led6, LOW);
delay(50);}

{digitalWrite(led7, HIGH);
delay(50);
digitalWrite(led7, LOW);
delay(50);}

{digitalWrite(led8, HIGH);
delay(50);
digitalWrite(led8, LOW);
delay(50);}

{digitalWrite(led9, HIGH);
delay(50);
digitalWrite(led9, LOW);
delay(50);}

{digitalWrite(led10, HIGH);
delay(50);
digitalWrite(led10, LOW);
delay(50);}

{digitalWrite(led11, HIGH);
delay(50);
digitalWrite(led11, LOW);
delay(50);}

{digitalWrite(led12, HIGH);
delay(50);
digitalWrite(led12, LOW);
delay(50);}

{digitalWrite(led13, HIGH);
delay(50);
digitalWrite(led13, LOW);
delay(50);}

{digitalWrite(led14, HIGH);
delay(50);
digitalWrite(led14, LOW);
delay(50);}

{digitalWrite(led15, HIGH);
delay(50);
digitalWrite(led15, LOW);
delay(50);}

{digitalWrite(led16, HIGH);
delay(50);
digitalWrite(led16, LOW);
delay(50);}

{digitalWrite(led17, HIGH);
delay(50);
digitalWrite(led17, LOW);
delay(50);}

{digitalWrite(led18, HIGH);
delay(50);
digitalWrite(led18, LOW);
delay(50);}

{digitalWrite(led19, HIGH);
delay(50);
digitalWrite(led19, LOW);
delay(50);}

}

mkovero, 2018-11-19:
thanks for this; INPUTs would require similar functionality, also sensor calibration would need some thought (pressure sensors, tps, temp sensors)