mkovero/7226ctrl #15: Read gear number from CAN.

Amarok73, 2019-04-12:
Hello Markus,

May I ask You, if You know how to read the gear number, lock-up state and gearbox oil temperature from the CAN (preferably through OBDII)?
I am using Jeep WG with this gearbox for around 3 years now, and since beginning I have idea to prepare Arduino based indicator of these values. 
I was trying unsuccessfully to sniff messages and analyze them, and when i lost my hope, there appeared new version of JeepDiagPro, that was showing these values. 
I know from Krzysztof Dymaniuk, that it wasn't so far in the scope of Your engineering, but perhaps as You already have all needed facilities it would be easier for You to investigate it?
I'd really appreciate if You can share anything substantial, as this fight with obsession i feel i am loosing slowly... :-)

astam1, 2019-04-23:
Doing this through the ODBII is going to be very difficult. Most likely your vehicle has a module that acts like a gateway from the OBDII to the internal can network(s). This is typically secured so that a sort of "code" is needed. Only standard OBD PIDs will be available via OBDII without the manufacturers "code".

I have an XK that uses the same transmission. Using a tap into the power train bus can tell you in the 06 XK the tcm transmits base 10 can ids 536, 824, 1048, 1312. At least in this application the current gear does not appear to be transmitted on the bus. Most of the messages are not decipherable. Code 824 contains the input shaft speed. In my vehicle there is no manual shift mode, only a mode that selected the highest available gear. The 1,2,3,4,5 displayed on the cluster is simply the highest gear available and is read from the shift control module. 

I have used an app called "obd jscan" which is able to get the atf temperature. It works through the obd port. You could possible get the obd2 "code" for your car my sniffing the obd2 port while running the app. At least when I last used it, it didn't show the current gear or lock-up state.

Lastly what is a jeep WG? Did you mean WK, WK2, XK, WK?

Amarok73, 2019-05-13:
Hello @astam1, thank You for Your comment. 
Jeep WG is the Grand Cherokee 2 released on European market. The WJ is the same but on the USA. 
I have tried couple of scanners, but so far, analyzing the dumps I wasn't able to find anything that could possibly be identified as the messages I am interested about, so I am affraid that these PIDs are only returning values instead of broadcasting these parameters.

astam1, 2019-05-17:
Alright so the vehicle is a diesel WJ with the 722.6? It sounds like you are using the OBDII port to try to get the current transmission gear. It is likely that the PID for current transmission gear is not supported on that vehicle (not supported on mine either). If this is the case there is no easy way to get the current gear. 

This information may be on the internal can bus between the TCM and ECU but you would have to splice into it to find out and it may not be available their either. If you are interested in the internal bus you may find [this](http://connectors.dcctools.com/home.htm) useful. If you type in your year and select WJ you can find the connector pinouts. Using this you can locate the can bus that you would need to tap into. Do this at your own risk though.

dvjcodec, 2019-05-19:
Gear number is coded in ID218 (I was read from Mercedes W210 CANbus network). ID218 is Transmission Control Unit. 
Sample message --> ID218 6 FE 20 11 40 AA 09
20 - is a shifter position (same like W202)
11 - gear number
40 - no kickdown / with kickdown is 44
AA - probably torque status, because is growing up when car accelerate and stop on AA or 64 when speed stable (doesnt matter what gear). 
I have a lot recorded CANbus messages from my car and will put all info on my Github
https://github.com/dvjcodec/Mercedes-W210-CAN-bus

Amarok73, 2019-05-19:
So far, I've found that my Arduino + CanBus Shield "sandwich" is ok (tested successfully on Volvo V40), so reason that it is not returning anything from OBD2 socket seems to be somewhere else. It may be, that CAN pins in socket are disabled - not connected.
I'll try to get CAN signal from radio harness, but still i have feeling, that documentation for these "features" in my Jeep model is really tough to find in the Internet.

dvjcodec, 2019-05-19:
Connect direct to TCU's wires. CAN-bus can't be disabled, because is diagnostic data for all systems in your Jeep WJ. If U have ELM327 U can test connection between OBDII and application. Use free diagnostic software (best for WJ 2.7CRD) --> https://jeepswj.com/software/. 
I will try to connect to my W163 tomorrow.

Amarok73, 2019-05-20:
Krzysztof, I mentioned this software in my first post, and yes - it is possible to read all interresting me parameters using this software from OBD2 with Bluetooth interface connected. So in theory it's not needed to connect into any harness, it's just matter of knowing how to read these values from OBD socket. Here are some screenshots from the paid version of the JeepDiag software.
[image]
[image]
[image]

dvjcodec, 2019-05-21:
CAN-bus data direct from Transmission Control Unit (EGS52)
53,752 218      8 CC 40 FF 48 13 80 FF 51 
53,769 338      8 FF FF 1F FF 00 FF 00 00 
53,771 418      8 20 20 82 80 FF FF 7F FF
Tomorrow I will conect to my MB W163 and give U info about where is current gear.

Amarok73, 2019-05-21:
Which one is the PID, 218, 338 and 418?

dvjcodec, 2019-05-23:
My ML do not have CAN pins in OBDII connector (probably WJ 2.7 same). CAN-bus is an internal network between modules, not for diagnostic. All data to ELM327 diagnostic (with current parameters and DTC's is delivered from SAE J1850 pins --> http://www.interfacebus.com/Automotive_SAE_J1850_Bus.html

UKSFM99, 2019-12-08:
Hate to comment on an old issue, but this is luckily possible, depending on what Model car you have.

I'm the author of this project: https://github.com/UKSFM99/W203-canbus, and have actually used gear number reading for 722.6 in the W211/203/209 Series. 

The following discoveries I've found on Bus B. Not Can C.

You should be able to read the gear number in CAN  PID 0x0003 (Its bits 12-16).

Additionally, Transmission fluid temperature can be read using PID 0x0222 (bits 0-8 aka Byte 1) within the frame. Doesn't seem to work on my W203 but does on my friends W211.

I am currently trying to reverse engineer Canbus C manually (Isolating the TCM to see what PIDS it sends). Will update if i find anything :)  

Hope this helps!

kacperziolo, 2020-02-11:
> Gear number is coded in ID218 (I was read from Mercedes W210 CANbus network with 722.6 transmission). ID218 is Transmission Control Unit.
> Sample message --> ID218 6 FE 20 11 40 AA 09
> 20 - is a shifter position (same like W202)
> 11 - gear number
> 40 - no kickdown / with kickdown is 44
> AA - probably torque status, because is growing up when car accelerate and stop on AA or 64 when speed stable (doesnt matter what gear).
> I have a lot recorded CANbus messages from my car and will put all info on my Github
> https://github.com/dvjcodec/Mercedes-W210-CAN-bus
> On my Git also find CANbus messages to read coolant temperature, rpm's and speed.
> 
> Mercedes W210 do not have ODBII standard, but maybe same CANbus coding is keep. In free time I will check in my MB W163 with 2.7CDI and 722.6 (same setup like jeep WJ).

Hi Krzysztof, could you share more informations about W210 AT CAN communication by any chance ? I'm swapping OM613 engine with 722.6 AT to 85' G-class and I need to make selected gear indicator for "tip-tronic" mode. Your data should help me a lot. Thanks you in advance

rnd-ash, 2020-02-15:
your mostly correct there!

I have decoded a Xentry SGS file which shows ALL the IDs stored within the 722.6 controller, there appears to be way more there than what you found out originally.

see here: https://github.com/rnd-ash/W203-canbus/blob/master/DAT_TRANSLATOR/DECODED/w203_w209%20CAN%20C%20ENGLISH.txt

and find the ECU called GS_218h as that's the ID message you found.

note though that all the GS_ ECU can messages are broadcast from the 722.6s controller, so you'll need to implement all of them.

I have opened a second issue here where I've dumped all the msg Params from the controller.

regards