mkovero/7226ctrl #18: Display pin out

captncockwash, 2019-12-25:
I’m trying to use a 7 pin SPI OLED. Could somebody please clarify the pin function for ‘DISPLAY 1’ to ‘DISPLAY 5’? Clk, MOSI, DC etc. 

I’ve looked at the U8G2 stuff but fear I’m just going round and round in circles. Should the display boot as soon as power is applied to the board. 

I’m a relative noob when it comes to the code. 

Thank you in advance.

mkovero, 2020-02-27:
Are you referring to some pcb pinout or? which revision?

asaner20, 2021-12-22:
I have a similar question. I soldered the board from this github, revision K. I would like to know the pinout for
connect 7pin SPI OLED display. Without the ability to see data in real time, I cannot check the controller. I also have a question, maybe you have a PC program to configure the controller. I am a car enthusiast, I have knowledge of a mechanic, but in programming I am a noob