mkovero/7226ctrl #19: ALL 722.6 TCU CAN MESSAGES

rnd-ash, 2020-02-13:
I found these ID's from a python script I wrote to decompile data found in Xentry:
https://github.com/rnd-ash/W203-canbus/blob/master/DAT_TRANSLATOR/parse_dat.py

These ID's are broadcast on CANBUS C - The ECU name is just the name of the message in Xentry, but they all come from the GS controller (AKA TCU) - Original text was in German so most of it has been translated :)

The offset and len in the messages are viewed from a big endian position within the can frame, where 0 is byte 0 and 64 is the 8th bit in the 8th byte

ECU NAME: GS_218h, ID: 0x0218. MSG COUNT: 31
	MSG NAME: MTGL_EGS - Motormomentenanf. Toggle 40ms +-10, OFFSET 0, LENGTH 1
	MSG NAME: MMIN_EGS - Engine torque requirment Min, OFFSET 1, LENGTH 1
	MSG NAME: MMAX_EGS - Engine torque requirment Max, OFFSET 2, LENGTH 1
	MSG NAME: M_EGS - Current engine torque, OFFSET 3, LENGTH 13
	MSG NAME: GZC - Target gear, OFFSET 16, LENGTH 4
	MSG NAME: GIC -Current gear, OFFSET 20, LENGTH 4
	MSG NAME: K_S_B - torque converter slipping, OFFSET 24, LENGTH 1
	MSG NAME: K_O_B - torque converter open, OFFSET 25, LENGTH 1
	MSG NAME: K_G_B - torque converter closed, OFFSET 26, LENGTH 1
	MSG NAME: G_G - off road gear, OFFSET 27, LENGTH 1
	MSG NAME: GSP_OK - gear shift program o.k., OFFSET 28, LENGTH 1
	MSG NAME: FW_HOCH - Driving resistance too high, OFFSET 29, LENGTH 1
	MSG NAME: SCHALT - Schaltung, OFFSET 30, LENGTH 1
	MSG NAME: HSM - Manual shift mode active, OFFSET 31, LENGTH 1
	MSG NAME: GET_OK - Transmission ok, OFFSET 32, LENGTH 1
	MSG NAME: KS - Knallstart, OFFSET 33, LENGTH 1
	MSG NAME: ALF - start enabled, OFFSET 34, LENGTH 1
	MSG NAME: GS_NOTL - Gearbox in limp home mode, OFFSET 35, LENGTH 1
	MSG NAME: UEHITZ_GET - Gearbox overheating, OFFSET 36, LENGTH 1
	MSG NAME: KD - Kickdown, OFFSET 37, LENGTH 1
	MSG NAME: FPC_AAD - Driving program for AAD, OFFSET 38, LENGTH 2
	MSG NAME: MPAR_EGS - Motor torque request parity (even parity) OFFSET 40, LENGTH 1
	MSG NAME: DYN1_EGS - Engagement mode / drive torque control, OFFSET 41, LENGTH 1
	MSG NAME: DYN0_AMR_EGS - Engagement mode / drive torque control OFFSET 42, LENGTH 1
	MSG NAME: K_LSTFR - Torque converter lockup free of load, OFFSET 45, LENGTH 1
	MSG NAME: MOT_NAUS_CNF - Confirm bit OFFSET 46, LENGTH 1
	MSG NAME: MOT_NAUS - Engine emergency stop, OFFSET 47, LENGTH 1
	MSG NAME: MKRIECH - creep torque (FF for EGS + CVT), OFFSET 48, LENGTH 8
	MSG NAME: FEHLPRF_ST - Status error check, OFFSET 56, LENGTH 2
	MSG NAME: CALID_CVN_AKT - CALID/CVN transmission active OFFSET 58, LENGTH 1
	MSG NAME: FEHLER - Error number counter, OFFSET 59, LENGTH 5
ECU NAME: GS_338h, ID: 0x0338. MSG COUNT: 2
	MSG NAME: NAB - Gearbox output speed (only 463/461, otherwise FFFF), OFFSET 0, LENGTH 16
	MSG NAME: NTURBINE - Turbine speed (NAG1/NAG2 only), OFFSET 48, LENGTH 16
ECU NAME: GS_418h, ID: 0x0418. MSG COUNT: 17
	MSG NAME: FSC - Driving position, OFFSET 0, LENGTH 8
	MSG NAME: FPC - Driving program, OFFSET 8, LENGTH 8
	MSG NAME: T_GET - Gearbox oil tempurature, OFFSET 16, LENGTH 8
	MSG NAME: ALLRAD - All wheel drive?, OFFSET 24, LENGTH 1
	MSG NAME: FRONT - Front wheel drive [1], Rear wheel drive [0], OFFSET 25, LENGTH 1
	MSG NAME: SCHALT - Schaltung, OFFSET 26, LENGTH 1
	MSG NAME: CVT - Stepless gears[1], Stepped gear [0], OFFSET 27, LENGTH 1
	MSG NAME: MECH - Gear mechanism varient, OFFSET 28, LENGTH 2
	MSG NAME: ESV_BRE - Apply brake when switched on, OFFSET 30, LENGTH 1
	MSG NAME: KD - Kickdown, OFFSET 31, LENGTH 1
	MSG NAME: GZC - Target gear, OFFSET 32, LENGTH 4
	MSG NAME: GIC - Actual gear, OFFSET 36, LENGTH 4
	MSG NAME: M_VERL - Torque loss, OFFSET 40, LENGTH 8
	MSG NAME: FMRADPAR - wheel torque (even parity), OFFSET 48, LENGTH 1
	MSG NAME: FMRADTGL - Factor wheel torque toggle 40ms + -10, OFFSET 49, LENGTH 1
	MSG NAME: WHST - Gear selector lever posititon (NAG, KSG, CVT), OFFSET 50, LENGTH 3
	MSG NAME: FMRAD - Wheel torque factor (FF for KSG), OFFSET 53, LENGTH 11

mkovero, 2020-02-27:
woot, thank you very much for these, will look into this at some point

rnd-ash, 2020-03-11:
no problem! I'm trying to make my own 722.6 TCU for my W203 as a hobby project (not using this code, sorry), but I thought I'd share those :)