mkovero/7226ctrl #20: Possible collaberation on updated controller for 2001-2011 Mercs

rnd-ash, 2020-09-19:
Hi.

As you know, I posted all the 722.6 CAN messages to another issue on here. I would like to ask if you are up for collaborating on a more modern drop in replacement TCU for the 203-204 series vehicles (Basically anything with a 722.6 from 2001 to 2011).

I had some ideas on how to update your codebase for CAN compatibility, but it will also require PCB modifications to allow for CAN C Tx/Rx support.

I also wish to add adaptive learning based on a simple neural network that can run on the Teensy 4 board, and allow it to be trained in pure manual mode (M), the apply the learned values in both Sport, Comfort, or Agility mode (A is supported on all 722.6 transmissions, but it was only implemented on the 204 chassis). Due to it being on CANBUS, its possible to take into consideration a lot more sensors on the car before determining if it should shift. Just some of whats avaliable on CANBUS below:
1. Individual wheel RPM (all 4)
2. Steering wheel angle (Driver input)
3. Steering wheel angle (Wheel output)
4. Boost pressure
5 .Engine torque production (Nm)
6. ESP / ABS actuation values

In regards to determining shift pressure. I can make that user configurable in each mode (S/C/A), in fact i plan to enable configuration over CANBUS for my Headunit project, allowing me to adjust shift parameters from a touch screen within the car.

I've also updated the CAN messages below with data extracted from my MercedesUI project (Custom infotainment system for older mercs)...This contains additional parsed values for enum based data:
[7226_can.txt](https://github.com/mkovero/7226ctrl/files/5250090/7226_can.txt)

mkovero, 2020-11-21:
Excellent idea, can rx/tx already works but is restricted to single channel currently. We have been considering doing hardware / software redesign, so if you feel you are up to it you can create algorithm of your own that we can try. 
I would prefer pure c++ instead of arduino and use of FreeRTOS helpers so realtime priorisation can be made properly. Refrain from using any interrupts or hardware timers too.
I'm really fond myself of M7 so it's probably going to be used no matter what happens. :>

rusefillc, 2022-01-16:
``I would prefer pure c++ instead of arduino and use of FreeRTOS helpers so realtime priorisation can be made properly. Refrain from using any interrupts or hardware timers too.``

that sounds like rusEFI hybrid timer but we use ChibiOS :)